adafruit/Adafruit_CircuitPython_GPS #5: Error while using a UBLOX module

eighthree, 2018-04-28:
Was testing this unit on a Trinket running Circuitpython 2.x.

When GPS is still locking on this error occurs:
```
Traceback (most recent call last):
  File "code.py", line 71, in <module>
  File "adafruit_gps.py", line 109, in update
  File "adafruit_gps.py", line 201, in _parse_gprmc
TypeError: can't convert NoneType to int
```
As soon as the GPS locks on the error disappears and things work properly.

margaret, 2018-05-14:
I am going to work on this (pycon sprints)

tannewt, 2018-05-16:
Fixed by #6!

closed 2018-05-16 by tannewt

Andon-A, 2018-09-11:
I'm getting similar to the original issue here. Before the GPS is locked/warmed up/whatever terminology you want to use, I'm getting errors in `update`:

```  File "main.py", line 76, in <module>
  File "adafruit_gps.py", line 112, in update
  File "adafruit_gps.py", line 206, in _parse_gprmc
TypeError: can't convert NoneType to int```

However, I more often get this issue:

```Traceback (most recent call last):
  File "main.py", line 76, in <module>
  File "adafruit_gps.py", line 102, in update
  File "adafruit_gps.py", line 143, in _parse_sentence
UnicodeError:```

As with the original issue, once the device begins functioning, it rarely gives any errors.

A quick glance at the indicated code for the NoneType error shows that it looks like it's attempting to error check the time_utc after attempting the conversion to int. I've taken some pokes at working with this, but my device isn't cooperating and it's midnight.

dhalbert, 2018-09-11:
@Andon-A I've seen the `UnicodeError` before: I think I was getting garbage on the serial line, but I can't remember. Could you just print out the input and take a look at it?

Is this Adafruit's Ultimate GPS or some other module?

Andon-A, 2018-09-12:
@dhalbert I'll fiddle tomorrow to get it to print the input.

It's a uBlox NEO-5Q-0-002

MBottinIUT, 2019-03-28:
Hi,

I just try to test a NEO-6M GPS module with CircuitPython on Itsy Bitsy M4 express board.
And when i put the simple test example into my main.py, I obtain the same TypeError like above.
The message is something like 'can't convert NoneType to int' (line 211 in '_parse_gprmc').

However, when I look at the oscilloscope, the signal is quite correct with an update every second.

I tried with Circuitpython v3.1.2 and v4.0.0 beta 5 and I've got the same results.

Does anyone fix it ?

Thanks a lot.

dhalbert, 2019-03-28:
This seems more like an issue with the serial data coming from the GPS. If you create the GPS object with `debug=True` as the last argument, it will print out what it's getting from the GPS module before it tries to parse it. That will help a lot trying to debug this.

MBottinIUT, 2019-03-29:
Thanks a lot for your answer.
I should have just tested my GPS info with a serial terminal. I would have noticed that the frames, if they were still present, remained empty of information.
In fact, I wasn't getting enough reception from the satellite signals where I was. I moved around a little bit and everything is now working perfectly.
Thank you again