adafruit/Adafruit_CircuitPython_GPS #10: Clarify that longitude/latitude precision is reduced with this library

reportingsjr, 2018-11-04:
I just spent a couple of hours trying to figure out why I was only getting 4 decimals from my gps wing with this library only to realize it is because the calculations used in _parse_degrees cut off a serious amount of precision.

This library will output decimal degrees to four places which, according to [this wikipedia page](https://en.wikipedia.org/wiki/Decimal_degrees), gives a precision of ~10m which isn't great.

It would be nice if a way to get higher precision decimal degrees was added to this library, even if only for logging rather than doing calculations.

ladyada, 2018-11-04:
hiya - we dont see where it cuts off the precision - it looks like we just parse it as a float?

reportingsjr, 2018-11-04:
It is parsed as a float, but since floats in circuitpython are single precision and not double precision you lose data when it is cast as the float.

The data comes in as a string like "8451.4091" and the 51.4091 (the gps minutes data) is getting reduced to 51.41.

ladyada, 2018-11-04:
i dont think that's a floating point issue? on a feather M0, 51.4091 is stored & printed the right way

> >>> x = 51.4091
> >>> print(x)
> 51.4091

reportingsjr, 2018-11-06:
@ladyada It happens when the data, which is a string, is cast to a float.

I'm using an adalogger M0 feather. When I open the REPL (via Mu) and run the following snippet (which is what the gps library does) I see the precision loss:
>>> float("8051.4091")
Which returns 8051.41

I ended up replacing the _parse_degrees function with something jank-tastic so I could get 6 decimals of data out of the gps module.
https://gist.github.com/reportingsjr/65195f67e43f47d8a5200f3b6742d875

6 decimals is about .1m of precision (see the table on this page: https://en.wikipedia.org/wiki/Decimal_degrees) which isn't super useful, but since I was already trying to get more decimals out I figured why not throw in an extra digit of precision!

reportingsjr, 2018-11-06:
Just to provide an example of the difference here..
In this first image by firmware was just using the format function with the lat/lon from the latest release of this library a la '{} {}'.format(gps.latitude, gps.longitude)
[image]

Here is my firmware after I switched to my janky way of getting higher precision:
[image]

(pink line is gps data, brown lines are hiking trails)

ladyada, 2018-11-06:
interesting... 
```
>>> float("8051.4091")
8051.41
>>> float("51.4091")
51.4091
```

ladyada, 2018-11-06:
OK it seems like you could have 2.4 floatingpoint just fine, its the 4.4 thats :(
so do you want to try parsing both sides and then storing as int degrees and float minutes?

ladyada, 2018-11-07:
kk
@dastels you just did some GPS stuff, any thoughts?

dastels, 2018-11-07:
It worked fine for my purposes.

ladyada, 2018-11-07:
k well, @reportingsjr i think the tuple may be the best way to do it then!

reportingsjr, 2018-11-07:
Ok, I can create a pull request this weekend!

kattni, 2018-12-03:
@reportingsjr Hello! Thanks for the PR! We've requested some changes there. Could you take a look? Greatly appreciated!

dherrada, 2019-05-07:
Ok so I've done some testing. This appears to be an issue with circuitpython and not the gps module.

While the raw NMEA data does have 4 decimal places, the ten meter accuracy for the fourth decimal place only applies to when it's been converted into the normal [decimal degrees](https://en.wikipedia.org/wiki/Decimal_degrees) format (not really sure what the real name is). 

`float('8051.4091')` does in fact return `8051.41` (on the m0 and m4 running 3.1.1, and the m4 running 4.0.0-rc1), but this is not mirrored in output of the gps_simpletest.py file. I got numbers with 6 decimal places when I ran that file as code.py.

This seems to be an issue with the core circuitpython libraries, I'll check if there's already an issue with that, and if there isn't, I'll make one

dherrada, 2019-05-07:
@reportingsjr I would reccomend doing what they do in gps_simpletest.py
```
print('Latitude: {0:.6f} degrees'.format(gps.latitude))
print('Longitude: {0:.6f} degrees'.format(gps.longitude))
```
This does return 6 decimal places, which gives the desired accuracy of 0.1m.

closed 2019-05-07 by kattni