adafruit/Adafruit_CircuitPython_GPS #18: commands send with send_command() might be ignored by the GPS

MSube, 2019-04-10:
send_command() uses multiple write() operations to send the sentence and the GPS might not recognize the sentence if there is a delay in between the write operations. 
Each send_command should produce one PMTK001 message and an absent message indicates that the command was not recieved.
The code below may or may not output 4 PMTK001 messages after a soft reboot on an Feather M4 Express with an attached GPS breakout.
Also, the GPS keeps sending the additional messages if the third PMTK314 is not received.

Combining the writes into a single one helps.

`
import board, busio, adafruit_gps
uart = busio.UART(board.TX, board.RX, baudrate=9600, timeout=3)
gps = adafruit_gps.GPS(uart, debug=True)

gps.update()
gps.send_command(b'PMTK314,-1')
gps.update()
gps.send_command(b'PMTK314,1,1,1,1,1,1,0,0,0,0,0,0,0,0,0,0,0')
gps.update()
gps.send_command(b'PMTK314,0,1,0,1,0,0,0,0,0,0,0,0,0,0,0,0,0')
gps.update()
gps.send_command(b'PMTK220,1000')

for i in range(20):
    gps.update()
`

dherrada, 2019-05-09:
The only way I've gotten this to work is by doing a hard reboot. It also appears that they get condensed. I.e. (0,0,1) and (1,0,0) become (1,0,1). Not sure why this is happening, but I think it's somewhere in either the update or send_command functions.

dherrada, 2019-05-15:
Disregard my previous answers. 

About the main issue you were having; gps.update needs to be called A BUNCH OF times before it actually returns anything. When I was testing your file, it ran over 4,700 times before it could actually return something. I'd try putting the gps.update in a while True loop and that will almost certainly fix that problem.

Now, about the second issue you were having with the GPS sending the additional packages. Not sure why that's happening, but from what I've gathered from the documentation about the GPS itself, it's almost certainly not on Circuitpython's end. Has something to do with the default settings on the GPS module itself. This can be solved by putting a time.sleep() in-between each PMTK314 packets being sent.

dherrada, 2019-05-15:
Something like this might work better for what you want
```
import board, busio, adafruit_gps, time
uart = busio.UART(board.TX, board.RX, baudrate=9600, timeout=3)
gps = adafruit_gps.GPS(uart, debug=True)

gps.update()
gps.send_command(b'PMTK314,-1')
time.sleep(2)
while True:
    gps.update()
    if gps.update():
        break
gps.send_command(b'PMTK314,1,1,1,1,1,1,0,0,0,0,0,0,0,0,0,0,0')
time.sleep(2)
while True:
    gps.update()
    if gps.update():
        break
gps.send_command(b'PMTK314,0,1,0,1,0,0,0,0,0,0,0,0,0,0,0,0,0')
time.sleep(2)
while True:
    gps.update()
    if gps.update():
        break
gps.send_command(b'PMTK220,1000')

for i in range(20):
gps.update()
```

dherrada, 2019-05-15:
Addressed in #28. Closed

closed 2019-05-15 by dherrada