adafruit/Adafruit_CircuitPython_GPS #46: TypeError: can't convert NoneType to int

fvzeppelin, 2020-05-30:
When running `gps_simpletest.py` on my IstyBitsy M0 Express with Circuitpython 5.3.0, I get
```
  File "code.py", line 58, in <module>
  File "adafruit_gps.py", line 148, in update
  File "adafruit_gps.py", line 301, in _parse_gprmc
TypeError: can't convert NoneType to int
```

dherrada, 2020-06-01:
I'll take a look at this sometime today or tomorrow. I'm assuming this keeps happening, correct?

caternuson, 2020-06-01:
What GPS hardware are you using and how is it connected to the ItsyBitsy M0?

fvzeppelin, 2020-06-01:
Yes, the problem is still there.
I am using this kind of module:
[](https://www.aliexpress.com/item/32818565324.html)

This addition of a delay between updates seems to fix the problem in simpletest:
```
while True:
    # Make sure to call gps.update() every loop iteration and at least twice
    # as fast as data comes from the GPS unit (usually every second).
    # This returns a bool that's true if it parsed new data (you can ignore it
    # though if you don't care and instead look at the has_fix property).
    gps.update()
    time.sleep(1.0)

```

dherrada, 2020-06-01:
@fvzeppelin Yeah, when I was adding some stuff to this library a few months ago, I had to do a similar thing. I was able to use a sleep time of .05 iirc and still have it work correctly, but I've forgotten why exactly that happened, and why adding the time.sleep fixed it.

It's good to know that you were able to get it working with minimal modification on a NEO-7M, which I wan't actually aware was supported or had been previously tested.

fvzeppelin, 2020-06-01:
@dherrada yeah, well, not completely:
Have to add a `(data[0] == '')` check to all the `parse-xxx` methods to prevent runtime errors in e.g. the
`time_utc = int(_parse_float(data[0]))`
conversion.
Nevertheless, the .05 seconds seem to be fine :)

dherrada, 2020-06-01:
@fvzeppelin Ok. I'll take a closer look later today or tomorrow and see if this is specific to the NEO-7M or if this is the case for all gps modules

fvzeppelin, 2020-06-01:
Here is what works for me if it helps:
[0001-fixes.patch.txt](https://github.com/adafruit/Adafruit_CircuitPython_GPS/files/4712882/0001-fixes.patch.txt)

dherrada, 2020-06-01:
@fvzeppelin Great. Thanks!

dherrada, 2020-06-08:
@fvzeppelin Hey, I was out most of last week, so I'll be taking a look at this now. Thanks for your patience

dherrada, 2020-06-08:
@fvzeppelin Ok, so I'm not getting the same issues. However, if you'd like to make a PR with the changes you made to get yours working, that'd be greatly appreciated. If your changes still work on the officially supported GPS modules, then we can see about merging them. In my experience with this library, and working with GPS hardware, sometimes it just does random stuff for no apparent reason, so any stability improvements would be greatly appreciated.

dherrada, 2020-06-08:
@fvzeppelin It also occurs to me that I've only tested this with the Ultimate GPS Breakout (MTK3339 chipset) and I haven't tested with the mini gps breakout (MTK3333 chipset). I'd assume they'd be pretty similar, since they've got the same manufacturer and all, but it couldn't hurt to give it a go with a different one, so I'm gonna try to get my hands on one of those and test with it.

By all means continue with the PR if you'd like to, just know that it might take me a bit longer to review because it might take a bit for me to get the mini gps

fvzeppelin, 2020-06-08:
@dherrada I just created a pull request.

closed 2020-06-09 by fvzeppelin