adafruit/Adafruit_CircuitPython_GPS #53: _parse_degrees doesn't handle negative values correctly

lesamouraipourpre, 2021-02-12:
The values returned from latitude and longitude are wrong if the value is negative.

The following test code:
```
# test_gps_parse_degrees.py

import adafruit_gps

print(adafruit_gps._parse_degrees("5412.25"))
print(adafruit_gps._parse_degrees("-5412.25"))
print("These should be the same but negative.")
```

outputs:
```
%Run test_gps_parse_degrees.py
54.204166666666666
-53.5375
These should be the same but negative.
```

jepler, 2021-02-12:
Thanks for your report!

As indicated by the leading `_`, the function `_parse_degrees` is intended only to be used internally to the adafruit_gps module and only to parse numbers in the form `dddmm.mmmm`.

I am not familiar with all possible "GPS" sentences, but the ones I am familiar with never supply a leading "-".  Instead, they have a separate field which specifies "N"orth, "S"outh, "E"ast or "W"est.

If the module is failing to support some GPS sentence, please let us know more about the sentence and the device that is emitting it.

lesamouraipourpre, 2021-02-12:
You're correct. I was getting funky Westing values while testing, but I didn't take proper account of reflections that I was receiving in my location and was looking for a problem that didn't exist. If I'd compared, the produced location to my Garmin sooner, I would have realised.

Apologies.

closed 2021-02-12 by lesamouraipourpre

jepler, 2021-02-12:
No worries, thank you!