adafruit/Adafruit_CircuitPython_GPS #55: The CRC check doesn't catch all bad messages

lesamouraipourpre, 2021-02-22:
Because, the CRC value in the NMEA sentence is only 8-bit, it is fairly easy for a bad message to get past the CRC check.
By observation, with my GPS on the window it takes on average ~3 hours before a bad message crashes the program.

This is usually triggered by `_parse_int` and `_parse_float` functions and can happen elsewhere.

lesamouraipourpre, 2021-02-22:
I intend to fix this by making the `_parse_int` etc. methods always return either a valid value or `None` and never raise a `ValueError`.

I also intend to put checks into the `_parse_gpgll` etc functions which will fail quickly if the wrong number of comma separated parts has been received.

tannewt, 2021-02-23:
Would it be possible to fail in the same way as CRC if parse int or float fail? That way you still get an error but you only have one type of error to handle.

lesamouraipourpre, 2021-02-23:
I'm aiming to keep it so that the update() method behaviour stays the same, specifically
`Returns True if new data was processed, and False if nothing new was received.`
Returning `False` in the event of an unparseable sentence will match the behaviour of a CRC failure.

tannewt, 2021-02-23:
Ok, that sounds perfect! Thank you!

PR #57 (2021-03-08): Fail bad sentences that pass the CRC check
As the CRC is only 8 bit and bad sentences will get past it, this PR attempts to catch bad sentences by comparing them to an expected parameter type list after the CRC check. Fixes #55

The `_update()` method does the following:
1. When available read a complete sentence from UART/I2C. Return `False` if not available. (No change)
2. Check the CRC. Return `False` if it fails. (No change)
3. If the sentence is NOT from a GNSS talker, assume it is valid and return `True`.
4. Call a _parse_XXX method for the sentence type, eg. `_parse_rmc()` for RMC.
5. I [... cut]
commit 7a59234e (2021-03-08): Refactor the sentence parsing (Fixes #55)
Refactor the date and time parsing into a method
Add an instance variable self.mode_indicator
Add an instance variable self.magnetic_variation

Sentence Parsing:
1) Read the sentence from I2C/UART (No change)
2) Validate the CRC, else fail (No change)
3) Call _parse_[SENTENCE_TYPE](), if the sentence was from a GNSS, else return True
4) The data length is compared against it's expected length, else fail
5) These then call _parse_data(PSEUDO_SENTENCE_TYPE,DATA)
6) The data is converted into the expected parameter types, else fail

PSEUDO_SENTENCE_TYPE i [... cut]
closed 2021-03-08 by tannewt