adafruit/Adafruit_CircuitPython_GPS #56: The shortest NMEA sentence can be less than 32

lesamouraipourpre, 2021-02-25:
Currently in `_read_sentence()` the code waits for at least 32 bytes before processing
```
def _read_sentence(self):
    # Only continue if we have at least 32 bytes in the input buffer
    if self.in_waiting < 32:
        return None
```

The shortest sentence I can find referenced in a datasheet is 15 bytes long:
`$PMTK501,2*28<CR><LF>`

I can find no definitive shortest sentence, but theoretically, there could be a sentence 11 bytes long:
`$XXXXX*ZZ<CR><LF>`
where the talker/sentence_type or proprietary_message is the message and needs no extra data.

ladyada, 2021-02-27:
sure if ya like you can submit a PR to shorten it to 11 bytes, not that it isnt a 'minimum sentence' length thing, since usually multiple sentences get emitted at once.

jposada202020, 2021-04-24:
@lesamouraipourpre now that changes were, was this fixed? let me know thanks :)

PR #62 (2021-04-26): Change the minimum sentence length to 11
Closes #56
commit 2b98d77f (2021-04-26): Change the minimum sentence length to 11 (Closes #56)
closed 2021-04-26 by tannewt