adafruit/Adafruit_CircuitPython_GPS #74: AttributeError: 'bytes' object has no attribute 'decode'

jeremyfsu, 2021-11-13:
I'm using CircuitPython version 7.0.0 on the Feather M0 Lora board. Also using a Beitian BN-220 GPS. 
Getting this error when trying to parse GPS:
main.py output:
('GPGSV', '3,3,11,21,28,129,21,27,31,045,15,30,40,311,33')
Traceback (most recent call last):
  File "main.py", line 53, in <module>
  File "adafruit_gps.py", line 282, in update
  File "adafruit_gps.py", line 588, in _parse_gsv
AttributeError: 'bytes' object has no attribute 'decode'

I'd be grateful for some help,
Thanks!
Jeremy

jepler, 2021-11-13:
It appears that this incompatibility with low-spec boards was probably introduced together with addition of GSV GSA sentence parsing:
  * #54

@lesamouraipourpre not all circuitpython microcontrollers have `bytes.decode`.  A possible alternative, assuming the message is ASCII, is `"".join(chr(c) for c in talker)`

jepler, 2021-11-13:
`str(talker, 'ascii')` is more natural than the first alternative I suggested, and is already used.

tekktrik, 2022-06-03:
@caternuson sounds like this is fixed?

caternuson, 2022-06-03:
@tekktrik Yep, should be.

@jeremyfsu Please try updating to the 3.9.8 release of the library:
https://github.com/adafruit/Adafruit_CircuitPython_GPS/releases/tag/3.9.8
which should be available in the library bundle:
https://circuitpython.org/libraries

tekktrik, 2022-07-21:
Closing since it seems like the issue was resolved!

closed 2022-07-21 by tekktrik