adafruit/Adafruit_CircuitPython_GPS #75: Usage of float causes timestamp_utc to be imprecise

mrdalgaard, 2021-12-08:
NMEA timestamp is written as HHMMSS.sss, which is stored as float in the module.

Later it is converted to int which is inaccurate:
See example:
>> int(184105.000)
184104

This is then math'ed into the fields in time_struct.

Propose storing the NMEA sentence as string instead, and grabbing the sections of the sting directly.

tannewt, 2021-12-09:
Note that it isn't the int conversion that is the problem. Storing it as a float is the issue. Try printing the float version back. My guess is that you'll get:

```
>> 184105.000
184104.0
```

mrdalgaard, 2021-12-09:
> Note that it isn't the int conversion that is the problem. Storing it as a float is the issue. Try printing the float version back. My guess is that you'll get:
> 
> ```
> >> 184105.000
> 184104.0
> ```

I actually don't:

>>> 184105.000
184105.0
>>> int(184105.000)
184104

tannewt, 2021-12-10:
:exploding_head:

PR #76 (2021-12-10): Enhanced parsing of timestamp_utc
Closes #75 

Uses string parsing instead of math for generating timestamp_utc.
Avoids imprecise usage of float.

Sorry if i messed this up, its my first pull request.

Tested on Adafruit Feather Bluefruit Sense, and CP7.2.0-alpha.0.
I do not have any low-memory MCUs to test on such as the M0
closed 2021-12-10 by tannewt