adafruit/Adafruit_CircuitPython_GPS #78: Lag in receiving gps lat long values

nayanagg, 2022-03-31:
Hi,
I'm getting data lag in receiving data from GPS when using threading in the python script.

BAUD_RATE SETTING = 115200
UPDATE_RATE SETTING= 100 milisecond

caternuson, 2022-03-31:
Can you clarify what you mean by lag. What device are you running the code on? Is it only happening when using threading?

caternuson, 2022-03-31:
> lag as in when using threading starts in python code the GPS timestamp starts lagging behind from actual time 

By how much? How are you determining this?

nayanagg, 2022-04-01:
by comparing with the gps timestamp with raspberry pi timestamp

caternuson, 2022-04-01:
Are the two time stamps acquired at the same time in the same thread? Or in separate threads?

Is the difference milliseconds? Minutes? Days? Weeks?

Can you provide example code that demonstrates the issue?

nayanagg, 2022-04-01:
Both timestamp collections running in the same thread.
the difference in timestamp is shown in the image, GPS time is in UTC, and python DateTime is in IST.

[image]

nayanagg, 2022-04-01:
code :

def gps_int():
    global gps,latitude,longitude,speed,accuracy,progSock, client_sock, client_info, state,BUFF_SIZE
    uart = serial.Serial("/dev/ttyUSB0", baudrate=9600,timeout=10)
    gps = adafruit_gps.GPS(uart,debug=False) 
    gps.send_command(b"PMTK251,115200")
    uart = serial.Serial("/dev/ttyUSB0", baudrate=115200,timeout=10)
    gps = adafruit_gps.GPS(uart,debug=False) 
    gps.send_command(b"PMTK220,100")
    gps.send_command(b"PMTK300,100")
    gps.send_command(b"PMTK314,0,1,0,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0")
    while True:
        gps.update() 
        if gps.has_fix:
            print("=" * 40) 
            print("Fix timestamp: {}/{}/{} {:02}:{:02}:{:02}".format(gps.timestamp_utc.tm_mon,gps.timestamp_utc.tm_mday,gps.timestamp_utc.tm_year,gps.timestamp_utc.tm_hour,gps.timestamp_utc.tm_min,gps.timestamp_utc.tm_sec,))
            print(datetime.now())
            if gps.speed_knots is not None:
                #print("Speed: {} Km/Hr".format(int((gps.speed_knots) * 1.852)))
                speed = int(gps.speed_knots*1.852)
            if gps.horizontal_dilution is not None:
                #print("Horizontal dilution: {}".format(gps.horizontal_dilution))
                accuracy = gps.horizontal_dilution
            latitude = ("{0:.6f}".format(gps.latitude))
            longitude = ("{0:.6f}".format(gps.longitude))
        else:
            print("Waiting for fix...")
            latitude = 0.0
            longitude = 0.0
            speed = 0.0
            accuracy = 0.0
            time.sleep(1)

ladyada, 2022-04-01:
the date time we get from the GPS is whatever is emitted, there's nothing in the library that manipuates it. so maybe its threading or maybe your raspi has a timing offset. either way, we recommend not using threading, none of the examples use it and its the fastest way to solve your problem as we have no plans to research your issue

grgrant, 2026-09-03:
@FoamyGuy Could this be closed given the age, lack of followup and LadyAda's response above?