adafruit/Adafruit_CircuitPython_GPS #84: Fix Timestamp not working on GPS Featherwing

DJDevon3, 2022-07-02:
https://github.com/adafruit/Adafruit_CircuitPython_GPS/blob/583fbd01ef7a85859f8bde51a8e83a0c86fe52c0/examples/gps_simpletest.py#L74

Fix Timestamp isn't working for me. Always remains the same timestamp. Using this example with Ultimate GPS Featherwing on nrf52840 board on UART. Didn't change anything in the example, just ran it, everything else seems to work fine. Maybe issue with my RTC?

serial output. doesn't matter how long it runs the fix timestamp always remains the same
``` ========================================
Fix timestamp: 7/2/2022 20:21:18
Latitude: obfuscated degrees
Longitude: obfuscated  degrees
Precise Latitude: obfuscated  degrees
Precise Longitude: obfuscated  degrees
Fix quality: 2
# satellites: 10
Altitude: obfuscated  meters
Speed: 0.01 knots
Track angle: 192.84 degrees
Horizontal dilution: 1.11
Height geoid: obfuscated meters
========================================
Fix timestamp: 7/2/2022 20:21:18
Latitude: obfuscated  degrees
Longitude: obfuscated  degrees
Precise Latitude: obfuscated  degrees
Precise Longitude: obfuscated  degrees
Fix quality: 2
# satellites: 10
Altitude: 27.7 meters
Speed: 0.01 knots
Track angle: obfuscated  degrees
Horizontal dilution: 1.11
Height geoid: obfuscated  meters

DJDevon3, 2022-07-07:
Neradoc came up with a fix that includes a timezone variable. Sets the RTC and can offset easily with a timezone variable. This is much easier to hit the ground running at least with a GPS featherwing. Can confirm this code works great.

```py
# SPDX-FileCopyrightText: 2021 ladyada for Adafruit Industries
# SPDX-License-Identifier: MIT

# Simple script using GPS timestamps as RTC time source
# The GPS timestamps are available without a fix and keep the track of
# time while there is powersource (ie coin cell battery)

import time
import board
import busio
import rtc
import adafruit_gps

TIMEZONE = -4

uart = busio.UART(board.TX, board.RX, baudrate=9600, timeout=10)
# i2c = busio.I2C(board.SCL, board.SDA)

gps = adafruit_gps.GPS(uart, debug=False)
# gps = adafruit_gps.GPS_GtopI2C(i2c, debug=False)  # Use I2C interface

gps.send_command(b"PMTK314,0,1,0,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0")
gps.send_command(b"PMTK220,1000")

the_rtc = rtc.RTC()

def _format_datetime(datetime):
    return "{:02}/{:02}/{} {:02}:{:02}:{:02}".format(
        datetime.tm_mon,
        datetime.tm_mday,
        datetime.tm_year,
        datetime.tm_hour,
        datetime.tm_min,
        datetime.tm_sec,
    )

had_a_fix = False
last_print = time.monotonic()
while True:
    gps.update()

    if gps.timestamp_utc and not had_a_fix:
        print("Setting the time")
        the_rtc.datetime = gps.datetime
        the_rtc.datetime = time.localtime(time.time() + TIMEZONE * 3600)
        had_a_fix = True

    # Every second print out current time from GPS, RTC and time.localtime()
    current = time.monotonic()
    if current - last_print >= 1.0:
        last_print = current

        if not gps.timestamp_utc:
            print("No time data from GPS yet")
            continue
        print("==========================")
        # Time & date from GPS informations
        print("Fix timestamp: {}".format(_format_datetime(gps.timestamp_utc)))
        # Time & date from internal RTC
        print("RTC timestamp: {}".format(_format_datetime(the_rtc.datetime)))
        # Time & date from time.localtime() function
        print("Local time: {}".format(_format_datetime(time.localtime())))
    time.sleep(10)```

Neradoc, 2022-07-08:
I think the basic issue might be with using the GPS as a time source for the RTC. I don't know all the inner workings, but it looks like doing that causes the time functions to return time based on `gps.datetime`. But that value is apparently only updated when the GPS gets a timestamp from the satellites (when receiving a GLL type "sentence").
```py
import time
import board
import busio
import rtc
import adafruit_gps

uart = busio.UART(board.TX, board.RX, baudrate=9600, timeout=10)
gps = adafruit_gps.GPS(uart, debug=False)
gps.send_command(b"PMTK314,0,1,0,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0")
gps.send_command(b"PMTK220,1000")

rtc.set_time_source(gps)

while not gps.timestamp_utc: gps.update()

print(time.time())
time.sleep(5)
print(time.time())
```
Result, which is quite unexpected when using the `time` module.
```py
code.py output:
2271171509
2271171509
```
So, a better use of it is to not use `set_time_source` but instead use the local RTC and update the time reference when we get one (maybe not every time either, like we do with NTP to sync only every now and then). Even then there is no telling when the timestamp was received if the code spends a long time between reads of the uart buffer though.
Maybe clarify what the "time source" example is supposed to do ?

grgrant, 2026-09-03:
@foamyguy I now have an ultimate GPS breakout (not the FeatherWing but the same 1616D chipset and I can't reproduce this error when using the current adafruit_gps library so I believe we can close this issue.

closed 2026-09-09 by FoamyGuy