adafruit/Adafruit_CircuitPython_GPS #88: "ImportError: no module named rtc" when running gps_time_source.py

P2AK, 2022-09-23:
I'm trying to run gps_time_source.py on my Pico with the Adafruit Ultimate GPS module, but I get this error:

File "<stdin>", line 11, in <module>
ImportError: no module named rtc

[image]

Sorry if the solution should be obvious, but I'm quite the novice Python user.  I can't seem to find the rtc library needed to satisfy "import rtc" .

This is what my Raspberry Pi Pico files/folders look like.
[image]

Just as an aside, do I have to define which uart tx and rx pins I'm using?  I was able to read the raw output from the Adafruit Ultimate GPS using UART1 TX/RX GP4/5 which are the two pins I have it connected to.  I used the method outlined [here](https://forum.micropython.org/viewtopic.php?f=21&t=12061).

tekktrik, 2022-09-23:
It looks like you've downloaded Blinka and a few other things to your board mistakenly.  Blinka allows CircuitPython libraries to work on a Raspberry Pi (NOT the Pico), but it itself isn't CircuitPython.  You should follow [this guide](https://learn.adafruit.com/welcome-to-circuitpython/overview) for getting CircuitPython to set up on your microcontroller and downloading this library to try it out.  If you're still having trouble, feel free to reach out on the #help-with-circuitpython channel in the Adafruit Discord.  Good luck!

closed 2022-09-23 by tekktrik

Neradoc, 2022-09-23:
It looks like you are using blinka on Micropython.
Blinka does not implement the rtc module, you will want to look up how to set the time in Micropython, ask on their forums or other places.