adafruit/Adafruit_CircuitPython_GPS #89: How should the Adafruit Ultimate GPS be wired to the Raspberry Pi Pico?

P2AK, 2022-09-27:
Sorry if this should be obvious, but how should the Adafruit Ultimate GPS be wired/connected to the Raspberry Pi Pico in order for this to work?  Please humor me here as I'm trying my best to get this working so I can use it as a time source to run the 7 segment LED for a clock which I already have working, but my cheap RTC module drifts too much :(

[image]

closed 2022-09-28 by askpatrickw

P2AK, 2022-09-28:
I had already gone through the Adafruit Ultimate GPS product website and the documentation and have now once again reviewed it, but there is no mention of the Pico in that documentation.   Clearly a lot of work has gone into this code .  I'm assuming you are part of the Adfruit org since it's your Github username and it links to Adafruit's consumer website.  Why even a rudimentary wiring diagram would be left out of this code is beyond me.  It's really tough getting started with projects like this when you're not an electrical engineer and have no formal training.  It's doubly frustrating when the company you bought the product from, and who has written code specifically for the product, isn't willing to assist with proper documentation, so I guess this is thanks for nothing.  Your reply really seems to be at odds with Adafruit's altruistic goal of encouraging people to learn, but maybe it's just about profit in the end.

anecdata, 2022-09-28:
@P2AK Many of the folks you encounter on these issue comments are volunteers from within the community, including askpatrickw and me, not Adafruit employees. Issues are for bug reports. Support is best in the forums or on Discord, as askpatrickw mentioned. There are plenty of people who are willing to help you get your project working, and also to try to improve documentation and project resources. I suggest you focus your questions constructively in one of the support areas.