adafruit/Adafruit_CircuitPython_GPS #103: Speed in kilometers per hour?

Mane55, 2023-07-01:
There is only .speed_knots but I wish there were the SI unit meters/sec or more popular kilometers per hour in land traffic. Km/h is available from NMEA GPVTG message. I think its is only a line of code and enabling it in send command.

snkYmkrct, 2024-10-07:
It was definitely more than one line of code, haha, since the VTG message was not being parsed at all. 
Anyway, here it is  https://github.com/adafruit/Adafruit_CircuitPython_GPS/pull/113 :smile_cat:

closed 2024-10-09 by tannewt

Mane55, 2024-10-10:
Thanks for the solution.

BR
Mauno

LähetettyOutlook for Android<https://aka.ms/AAb9ysg>

Closed #103<https://github.com/adafruit/Adafruit_CircuitPython_GPS/issues/103> as completed.