adafruit/Adafruit_CircuitPython_GPS #114: Fix Quality 3d returns undocumented values

jar349, 2024-10-07:
The documented values for `fix_quality_3d` are as follows in the snippet below:
https://github.com/adafruit/Adafruit_CircuitPython_GPS/blob/41f15963f7bbb308064f20f22a69b7cc3238da41/adafruit_gps.py#L272-L279
However, the value I am getting for that attribute is `0`. If this is an expected value under certain conditions, would you mind updating the documentation to explain? If it is never expected for the value to be `0`, could you please explain what I may be doing wrong on my end to receive that value?  All other values I am getting look good/correct.

I suspect that the culprit is this block of code:
https://github.com/adafruit/Adafruit_CircuitPython_GPS/blob/3e13731caea98b731941fbdcb007a822ca0c38ae/adafruit_gps.py#L634-L642
If there's code beyond initializing it to `0` that sets it to `0` - as this block does - then `0` probably does have a semantic meaning that can be documented.

snkYmkrct, 2024-12-14:
@jar349  
The GPS module only receives the data (NMEA sentences) specified in its initialization commands. 
This is documented in the usage example code:  https://github.com/adafruit/Adafruit_CircuitPython_GPS/blob/3e13731caea98b731941fbdcb007a822ca0c38ae/examples/gps_simpletest.py#L31-L44  as well as in the referenced [GPS module datasheet](https://cdn-shop.adafruit.com/datasheets/PMTK_A11.pdf) -- see page 12 -- and various learn guides on the Adafruit page, like [this one](https://learn.adafruit.com/adafruit-mini-gps-pa1010d-module/circuitpython-python-uart-usage#gps-example-code-explained-2964457).

Since the `fix_quality` parameter is part of the GSA NMEA sentence, GSA data needs to be enabled in the initialization command, like so: `gps.send_command(b"PMTK314,0,0,0,0,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0")`. Otherwise, the behavior you're observing is correct, `self.fix_quality_3d` is `0` because there is no valid data to be parsed, and this is the default value. 
The "culprit" you're referring to is also correct behavior, since it only occurs when that particular GSA data packet is corrupted, and cannot be parsed. 

Usually it's better, and faster, to ask for library usage help on the [Adafruit forums](https://forums.adafruit.com/), or in the [Discord server]( http://adafru.it/discord ).

jar349, 2025-02-08:
> Usually it's better, and faster, to ask for library usage help on the [Adafruit forums](https://forums.adafruit.com/), or in the [Discord server](http://adafru.it/discord).

Thank you for that guidance. I have followed-up here: https://discord.com/channels/327254708534116352/537365702651150357/1337633627692859473